qualcomm-linux/meta-qcom-distro #29: Packagegroup bbappend is causing buildfailures

koenkooi, 2025-03-24:
From https://github.com/qualcomm-linux/meta-qcom/actions/runs/14031828445/job/39294390951:

```ERROR: packagegroup-qcom-1.0-r0 do_package_write_ipk: An allarch packagegroup shouldn't depend on packages which are dynamically renamed (libinput to libinput10)```

ndechesne, 2025-03-24:
Context: https://lists.openembedded.org/g/openembedded-core/topic/patch_4_4_package_add/87011325
However I think we should remove libinput anyway, no?

lumag, 2025-03-24:
> Context: https://lists.openembedded.org/g/openembedded-core/topic/patch_4_4_package_add/87011325 However I think we should remove libinput anyway, no?

Not quite, libinput is both a library and a set of udev rules. A possible option might be to split libinput in OE-Core, separating udev rules to a separate package and making everybody depend on udev instead of a lib package.

ndechesne, 2025-03-25:
> Not quite, libinput is both a library and a set of udev rules. A possible option might be to split libinput in OE-Core, separating udev rules to a separate package and making everybody depend on [...]

in Debian, the udev rules are in libinput-bin. but i am still not sure why we want to keep libinput at all in this packagegroup..