qualcomm-linux/meta-qcom-distro #53: Container runtime support

koenkooi, 2025-07-23:
It would be nice to have support for containers in your images. The question is which runtime to we pick: docker-moby, podman or something else.

lumag, 2025-07-23:
I'd vote for podman.

lool, 2025-07-23:
Debian images carry docker from the docker.io Debian package; I vote for Incus!

vkraleti, 2025-07-25:
No strong opinion but Qualcomm Linux 1.0 supports docker. May be it is good to algin.

lumag, 2025-07-25:
> No strong opinion but Qualcomm Linux 1.0 supports docker. May be it is good to algin.

Did you have any pro et contra when selecting docker for QLI 1.0?

ricardosalveti, 2025-07-30:
The normal user expectation we currently have is more aligned with 'docker' (part of qli downstream, foundries images, etc), so I would vote for that to be the default from the distro perspective.